kitschpatrol/brain #12: signal strength

cslinux, 2015-01-02:
I've got the mindflex 2 wired to arduino uno , but the signal strength never reached the zero and always the attention and meditation values are zeros, that's when i have the headset on my head, the lowest value reached for the signal strength was 26, any idea what's wrong?

Couby, 2015-01-02:
Did you try using conductive gel or paste for electrodes ?
I tried this with good results in the same situation.

cslinux, 2015-01-02:
Which electrodes you mean ( mindflex Tx & GND ) and how exactly?

steeph-k, 2015-01-02:
The electrodes are the metal knobs on the headband. You can just put the gel between the electrodes and the skin. Alternatively, saline solution also helps, but not for long. Ultrasonic gel might be a cheap alternative. I also like to use adhesive disposable electrodes sometimes. They give also better results than the cheap electrodes integrated in the headband. For long-term use you might want to think about  silver electrodes as a replacement of the integrated ones. Self-made ones from conductive silver fabric are quite comfortable to wear.

Couby, 2015-01-02:
@cslinux I use it on both.

@steeph-k 
> Self-made ones from conductive silver fabric are quite comfortable to wear.
Do you have any recipe or build log about such self-made electrodes ? Sounds cool :-)

steeph-k, 2015-01-02:
Oh, I guess I misunderstood the question about which electrodes. Is it called Tx sometimes?

There is this [guide to make replacement sensors for the discontinued Zeo headband](http://www.instructables.com/id/DIY-Replacement-Zeo-Sleep-Monitor-Headband-Sensor/). I think it should give a good idea how to make a similar headband.

I didn't make one myself (so far). But I have used a headband made after this guide and it was great! :)

cslinux, 2015-01-02:
If you can attach a picture showing these electrodes i'll really appreciate
it as well can i use paste on them?

steeph-k, 2015-01-02:
Or maybe I didn't misunderstand you. I only have an old Mindflex. It looks a bit different. And I couldn't find a picture of the new one where the electrodes are visible. There is only one electrode on the forehead. The other two are in the earclip. Depending on what you want to do it would be better to place at least one of the earclip electrodes on the sculp as well. But that's not a matter of signal strength, just a side note.

cslinux, 2015-01-02:
So what do think  ppl? Just i have to use a paste on the ear clip and the forehead electrode?

steeph-k, 2015-01-02:
I don't think that paste would be a good solution. But you don't have to buy gel, either. Have a look at this: http://www.bioshare.info/en/diygel

cslinux, 2015-01-03:
Actually i've tried with paste and salt/water , always signal strength isn't lower than 26 and always meditation/attention is zeros even on the processing brain graph-er.

steeph-k, 2015-01-03:
Shouldn't meditation and attention still go up sometimes even with a connection value of 26? If connection strength is the issue maybe one of the connections of the wires is damaged. Or maybe your body is very very dry. (Do you happen to be a zombie?) So it doesn't sound anymore like the connection of the electrodes is the source of the problem.

cslinux, 2015-01-03:
You mean the Tx / GND soldering?

steeph-k, 2015-01-03:
Maybe. If they look especially bad, I'd re-solder them. It's not very likely that this is it. But I don't have a better idea.

cslinux, 2015-01-03:
:( ... ppl any comment ???!! suggestions ?

sicklebrick, 2015-01-03:
Basically just adding an optoisolator did the trick for me.

I tried conductive gel, filing the electrodes, resoldering, using a salty
solution, all sorts of interference shielding, grounding etc and it didn't
work. I even rang up a neuroscientist for tips...

Bought this for a pound and it worked instantly, 90% of the time it doesn't
need the gel either.
http://www.maplin.co.uk/p/high-voltage-opto-isolator-ay44x
​
So I'm not sure if it's my PC specifically or my Uno, but since you should
probably surge-protect your brain anyway, I"d say go for it.

steeph-k, 2015-01-03:
Definitely a good tip. If you don't transmit the signal wirelessly adding an opto-coupler is a good idea. I don't know if it's enough to protect you from a shock if something goes very wrong, but it should improve the signal quality.

cslinux, 2015-01-04:
and this how can be wired to the mindflex or the uno ?

cslinux, 2015-01-04:
As well which of the following you meant ?

http://www.ram-e-shop.com/ds/ic/MOC3063.pdf

http://www.vishay.com/docs/83725/4n25.pdf

sicklebrick, 2015-01-04:
steeph-k: Heh, they're actually rated for about 5000v... kinda impressive
considering how easy they are to blow out.

cslinux: If I remember right, people have had success with the 4n25. For
wiring instructions, I'd suggest maybe finding a youtube example, or
checking google images for schematics.

cslinux, 2015-01-05:
sicklebrick: The 4N25 should be wired to the arduino or to the mindflex ??! , sorry for such a question .. :(

cslinux, 2015-01-05:
sicklebrick: i couldn't find any wiring instructions, as well i don't know if i have to wire it between the arduino and the Mindflex Tx / GND or what ? i.e the inputs should be the mindflex tx and then the optocoupler collector goes to the rx ?

cslinux, 2015-01-07:
ppl any comment?!!

steeph-k, 2015-01-08:
>Mindflex TX&RX -> Optocoupler transmitter side -> Opocoupler reciever side
-> Arduino TX&RX

Shouldn't there be one for Tx and one for Rx?

sicklebrick, 2015-01-10:
Nah sorry, I was tired and not thinking straight... you don't use the
mindflex's RX at all and don't use the arduno's TX at all. It doesn't
transmit anything back along the TX line!

In theory you could do it with an infrared LED and detector at the other
side, but that kinda setup won't handle the transmission speed properly...
so coupler =)

Also, I found with the second set of couplers I bought that the mindflex
was a little short on power, so I used the TX line with a transistor and
4.5v supply on the transmit side...
Oh, and a little pull up resistor on the receive side.

Pic:
http://i.imgur.com/OtV26Kg.png?1

cslinux, 2015-01-12:
which kind of transistor and how it's connected in details? i see a capacitor and a resistor on the transmit side and a resistor on the receive side.

cslinux, 2015-01-12:
As well i've this ECG electrode http://store.fut-electronics.com/ecg-electrode/dp/1188 , but not sure if it can help in the signal strength issue and if yes how to connect it to the Mindflex electrodes

steeph-k, 2015-01-13:
To properly connect these electrodes you need a cable like the one suggested by the shop: http://store.fut-electronics.com/ecg-electrode-cable-%283-wires%29/dp/1437

Cut off the jack plug and solder the wires directly to the ThinkGear module.

cslinux, 2015-01-13:
steeph-k: which ThinkGear module, i'm using Mindflex and it has neurosky module with TX and RX already connecting the Tx to Arduino Rx and GND to GND

steeph-k, 2015-01-13:
Yes, that's the TGAM (ThinkGear ASIC Module) from NeuroSky.

cslinux, 2015-01-13:
steeph-k: so at the time i have the arduino connected to its Tx , so where i have to solder the jack plug?

steeph-k, 2015-01-13:
I wouldn't use the jack plug at all. The cable is too long for this use case. But you can use a jack socket if you want, just not a very cheap one.

I don't know by heart which contact is which. But they are labelled. Or you can look them up in the TGAM spec sheet (http://wearcam.org/ece516/neurosky_eeg_brainwave_chip_and_board_tgam1.pdf) or the datasheet (http://www.seeedstudio.com/document/pdf/TGAM%20Datasheet.pdf).

cslinux, 2015-01-13:
i'll solder the wires but to the tx or the EEG and ground electrode ?

steeph-k, 2015-01-13:
The electrodes have to go to the electrode connections ("EEG Electrode", "Ground Electrode" and "Reference Electrode" in the spec sheet), not the serial connection (Tx, Rx).

cslinux, 2015-01-13:
do you think i still have to use an optocoupler ?

cslinux, 2015-02-03:
sicklebrick:  Can you tell how should an opto-coupler improve the signal quality at the time i think it isolates 2 circuits from each other power wise lets say...

sicklebrick, 2015-02-09:
1- It will prevent any power surge from coursing through your juicy
conductive head.
2- It prevernts a ton of noisy interference from your USB
port/PC/monitors/blah.

It's a one-way deal, yeah, but it allows the transmission from the mindflex
to the ardunio via an infrared LED and detector.
It's kinda like putting an LED on the end of the mindflex's transmit... and
a detector on the arduino.. so instead of pulsing the values in, it flashes
them in... the chip just has a tiny LED and detector in it.
The reason you use an optocoupler instead of an actual LED though is
because of the response time... a regular one may not be fast enough (like
the kind in a TV remote)... but it's the same thing..  no physical
connection.

I'm not going to go over how to use one, cause I'm not taking
responsibility if you screw everything up, but this book will get you up to
speed quite quickly (well written- I understood most of it before
highschool):
https://www.google.co.uk/search?q=forrest+m+mims+getting+started+in+electronic

Seriously, grab a PDF or something, read it through and get the thing
hooked up. It's awesome fun!

eried, 2015-03-04:
Same issue here, quality is always >26 (25 touching the metal pad with my tongue) but attention and meditation is always 0. Any other hint? maybe is a device failure?

sicklebrick, 2015-03-05:
I had similar issues actually... was working maybe half the time or less
even with the opto coupler so I used a transistor + external 3v (Double A
batteries) power supply to feed the signal into the optocoupler. Now it
works ~90% of the time without any cleaning or gel.

Would've mentioned it sooner, but so many people seem to be struggling with
the idea of the opto isolator itsself that I didn't want to overcomplicate
things..

I can't remember the polarities off hand, and have to run to work, but
something like this should do the trick if you have 5-10 mins to fiddle
about. http://i.imgur.com/Hfvlq1J.png

Heheh, good luck with that image.

cslinux, 2015-03-05:
Which kind of transistor , as well do you mind sharing a schematic for the optoisolater wiring?

eried, 2015-03-06:
Hey sicklebrick thanks for the image, I guess the transistor could be any common one like the 2222N2 (probably?), but my question is why is this needed? just because noise induced by the arduino? I am using the 3xAAA from the MindFlex (only wired TX and GND)

sicklebrick, 2015-03-06:
Mmhh yeah, I guess most transistors would do but I think that's what I used
(actually hunted around for the circuit earlier but it's packed away
somewhere )

I think it's required for 2 reasons overall.. (just a guess based on
observations here)

1- Noise on the ground line from your PC and its USB ports getting into the
heart of the EEG and screwing with the fine measurements.
I asked a neuroscience PHD student about this and he suggested all manner
of fixes, but it seemed to be the PC itsself causing issues. Strangely
enough it happens on my laptop running only on batteries.

2- The TX line doesn't have enough power to drive the infrared LED inside
the optocoupler.

Although the symptoms could suggest other issues I guess - especially for
example the laptop being just as noisy... it might just be that I've
grabbed the wrong transistor and inverted the signal (though I'd expect
that to generate some heat), or some other thing I haven't considered or
even heard of. Point is, the 'circuit' I posted earlier does get the job
done and the signal has been good enough to actually control a few things
with on the PC end. (in a very basic way).

I guess since the neurosky board runs on batteries anyway you could use the
same batteries with a third wire providing 3V to the transistor/coupler,
but maybe do that once you've got the rest of it working.

When it does start working, and you get those good, strong, clean signals
it's pretty awesome, worth a day or two of fiddling and the $5 in parts you
might have to pay on top.

zombodotcom, 2015-07-26:
Hey, I got a mindflex duel, so 2 headsets, I'm only testing one headset so far, I turn the headset on and I can get data for a few seconds on the braingrapher with an arduino, then I'm getting 200,0,0 some other guy had the same question and I asked him on twitter with one question and he blocked me for some reason? 
I get a few seconds of data every time I turn it on and off again then it cuts out to 200,0,0

Any ideas ? I'll try the other headset when I have time

cslinux, 2015-11-24:
sicklebrick:  I've tried to connect the EEG module (TGAM1)  to arduino through a 4N25 Optocoupler, the connection without the coupler works fine, but when used to connect the EEG Tx to optocoupler pin1 , EEG GND to pin2 , then arduino RX to pin5 and Arduino GND to pin4, it doesn't work, asking for any suggestion or advices on what can be the issue?

sicklebrick, 2015-11-30:
Woops, sorry for the late reply - You tried powering the opto with another
battery+transistor?

ancymathai, 2015-12-17:
it is the project shown here : http://frontiernerds.com/brain-hack 
 when it is time to start getting data in serial monitor it just shows nothing...
i double checked the connections i soldered and everything is good...
the code is this :

// Arduino Brain Library - Brain Serial Test

// Description: Grabs brain data from the serial RX pin and sends CSV out over the TX pin (Half duplex.)
// More info: https://github.com/kitschpatrol/Arduino-Brain-Library
// Author: Eric Mika, 2010 revised in 2014

#include <Brain.h>

// Set up the brain parser, pass it the hardware serial object you want to listen on.
Brain brain(Serial);

void setup() {
    // Start the hardware serial.
    Serial.begin(9600);
}

void loop() {
    // Expect packets about once per second.
    // The .readCSV() function returns a string (well, char*) listing the most recent brain data, in the following format:
    // "signal strength, attention, meditation, delta, theta, low alpha, high alpha, low beta, high beta, low gamma, high gamma"    
    if (brain.update()) {
        Serial.println(brain.readErrors());
        Serial.println(brain.readCSV());
    }
}

richardbeattie, 2016-05-01:
Hi all I am having this problem as well, so I purchased a opto-isolator(http://www.maplin.co.uk/p/high-voltage-opto-isolator-ay44x) however I am stuck on its wiring, do I just soldering the wire from the T pin and the ground Pin to it or is there another wire that has be attached as well

Thanks 
Richard

SeverinPaar, 2020-11-15:
Using my Laptop instead of my PC worked, thanks!

johndderry, 2020-12-13:
Hi everyone I just make my setup with a mindflex, and I get the same results in that the signal doesn't get below 26 and bounces around between that and 52 or something. ONCE it did go to zero and sure enough attention and mediation values began to show. 
What I have learned is this: leaving the headset in the same place and unplugging the arduino usb with the serial connection intact, I turned on the mindflex base unit was able to calibrate and control the lights as normal. But when I plugged the arduino back in, the base unit immediately went back to calibration mode, AND COULD NOT CALIBRATE. Unplug the arduino, everything is ok.
I think the presence of the connection (voltage?) to the arduino RX line is interfering with the operation of the serial connection from the daughter board, presumable because it is paralleled with the headset's native controller. I am considering disconnecting the daughter board's connection to the native controller and trying it. 

John Derry

johndderry, 2020-12-13:
No, it's not the serial connection. With just the ground wire connected alone, whenever the arduino is running the mindflex is not working properly. I also tried an external 9v power source with no usb connection; same thing,

sixtyfive, 2020-12-14:
Perhaps try what @r-bt did and go with an optocoupler?

johndderry, 2020-12-15:
Yes I will try that. Thanks!

closed 2023-11-19 by kitschpatrol