kitschpatrol/brain #13: Use of the library.

ybucherenick, 2015-07-08:
Hello, I need to use this library in C # because after obtaining the wave's I have to use a neural network but as you know C # does not accept .h but .dll, which has given me problems since I can not generate the .dll , I can not even generate the .h as it gives me error when you compile the cpp. Not if you can help me with this, I would be very grateful.
[image]

kitschpatrol, 2015-07-08:
Hello @ybucherenick, unfortunately this project targets the Arduino platform, and likely isn't trivially compilable into a DLL.

If you don't have to use "hacked" hardware, I recommend looking at the [ThinkGear Communications Driver](http://developer.neurosky.com/docs/doku.php?id=thinkgear_communications_driver_tgcd) DLL which is advertised to play nicely with C#.

Good luck!

closed 2015-07-08 by kitschpatrol

ybucherenick, 2015-07-08:
Thanks,

I need to use the hacked hardware because i'm using c# to my projects and i can't change it to c++; is there any way to make this library as .dll?. 

Also i can't compile the code, the #include "Brain.h" keeps throwing an error and it doesn't generate the .h file

sicklebrick, 2015-07-09:
Are you trying to compile the brain library for windows?
That's supposed to be compiled within the "Arduino" software and uploaded
to the board.
From there it'll dump the data via a virtual serial port to your PC ( like
COM6 or whatever ), and you can read it via the serial monitor.

Good news is super easy to read the values from a COM port in C# (Visual
Studio anyway, can't be sure with Mono/Unity, etc).
Off the top of my head, you drop a component into your project via the
designer, tell it which port/baud rate, and don't worry about the low level
implementation.
(Just remember to close the one in the Arduino software first)

ybucherenick, 2015-07-09:
Thanks for the information! I do what you tell me in C #.