kitschpatrol/brain #15: Incorrect packet headers - no SYNC

bcaller, 2015-11-01:
Using a Mindflex.
I can read from the T pin. However, the Brain library doesn't recognise the data.
Connecting the R pin to the Arduino prevents the ThinkGear from outputting data.
Have I messed up the soldering (quite possible).
At 9600 baud I can't see two 0xAA bytes.
Please help.
Here is the data from the first few seconds (I wasn't wearing the headset):

[0x0]
[0x80]
[0x80]
[0x80]
[0x25]
[0xD]
[0x0]
[0x65]
[0x0]
[0x80]
[0x0]
[0x80][0x1C][0x80][0x80][0x80][0x6][0x80][0x80][0x48][0x2][0x4C][0x4A][0x70][0x80][0x0][0x5][0x0][0x80][0x80][0x53][0x0][0x5][0x0][0xC4][0xAA][0x5][0x2][0x5D][0xB][0x2][0x8A][0x0][0x2][0x19][0xB][0x2][0x11][0x1][0x0][0x80][0x80][0x80][0x5][0x2][0xF0]
[0xAA][0x25]
[0xD]
[0x10]
[0xE7][0x70]
[0x0]
[0x2][0xA0][0x5]
[0x0]
[0x0]
[0x80]
[0x6][0x0][0x74][0x80][0x0][0x80][0x80][0x80][0x0][0x80][0x80][0x6][0x0][0x81][0x80][0x0][0x80][0x0][0x0][0x80][0x80][0x0][0x0][0x0][0x0][0x5][0x0][0x56][0x3][0x2][0xF0]
[0xAA][0x25]
[0xD]
[0x18]
[0x45]
[0xEE][0x14]
[0x0]
[0x0]
[0x55]
[0x80]
[0x0]
[0x80][0x0][0x0][0x6][0x0][0x0][0x80][0x0][0xB0][0x2][0xB1][0xCD][0x0][0x80][0x80][0x0][0x5][0x0][0x30][0xAA][0x5][0x2][0x1][0x85][0x0][0x80][0x80][0x80][0x5][0x2][0x1][0x85]
[0x0]
[0x80]
[0x80]
[0x80]
[0x25]
[0xD]
[0x24]
[0x0]
[0x80]
[0x9B][0x28]
[0x80][0x80][0x80][0x80][0x80][0x6][0x0][0x89][0x3][0x80][0x1][0x0][0x0][0x0][0xA4][0x4][0x5][0x0][0xB8][0xAA][0x5][0x2][0x70][0x61][0x0][0x80][0x80][0x80][0x5][0x2][0x1][0x85][0x0][0x80][0x80][0x80][0x5][0x2][0x1][0x85]
[0x0]
[0x80]
[0x80]
[0x80]
[0x25]
[0xD]
[0x34]
[0x80]
[0x80]
[0x0]
[0x0]
[0x0]
[0x80][0x84][0x0][0x80][0x80][0x5][0x80][0x0][0x1D][0x96][0x20][0x0][0x0][0x8][0x1E][0x0][0x80][0x80][0x80][0x80][0x4][0x5][0x0][0x4F][0x3][0x2][0x1][0x85]
[0x0]
[0x80]
[0x80]
[0x80]
[0x25]
[0xD]
[0x24]
[0x0]
[0x80]
[0x80]
[0x26][0x6][0x0][0x80][0x80][0x18][0x7][0x0][0x80][0x0][0xAA][0x1D][0x80][0x80][0xC2][0x6][0x0][0x6][0x0][0xF9][0x6][0x80][0x80][0x0][0x5][0x5][0x0][0x41]
[0x0]
[0x80]
[0x80]
[0x80]
[0x25]
[0xD]
[0x34]
[0x0]
[0x0]
[0x80]

steeph-k, 2015-11-01:
This indeed looks very different from what it should look. http://wearcam.org/ece516/mindset_communications_protocol.pdf
I don't think the TGAM has a mode that is supposed to output data like this.
I can imagine that connecting two unrelated points with too much solder could cause something like this. Did you change the data mode by soldering the bridges? These are tiny. You really need a small soldering tip, a magnifier and a steady hand for them. I messed up these once (irreversibly), but it works as intended even with soldering points ripped off.

bcaller, 2015-11-01:
Thanks so much for your reply.
I thought again about "Connecting the R pin to the Arduino prevents the ThinkGear from outputting data.". I noticed that the output from R was the same as from T so I cleaned up my soldering with copper. It appears that there was a rogue connection between T and R.
It's now working.
Getting a number of packet length and checksum errors but I'm ok with that for now.
One note... the max packet length should be 169 rather than 32 in Brain.h?

closed 2015-11-01 by bcaller