kitschpatrol/brain #16: problem with arduino brain library serial monitor

ancymathai, 2015-12-17:
it is the project shown here : http://frontiernerds.com/brain-hack 
when it is time to start getting data in serial monitor it just shows nothing...
i double checked the connections i soldered and everything is good...
the code is this :

// Arduino Brain Library - Brain Serial Test

// Description: Grabs brain data from the serial RX pin and sends CSV out over the TX pin (Half duplex.)
// More info: https://github.com/kitschpatrol/Arduino-Brain-Library
// Author: Eric Mika, 2010 revised in 2014

#include

// Set up the brain parser, pass it the hardware serial object you want to listen on.
Brain brain(Serial);

void setup() {
// Start the hardware serial.
Serial.begin(9600);
}

void loop() {
// Expect packets about once per second.
// The .readCSV() function returns a string (well, char*) listing the most recent brain data, in the following format:
// "signal strength, attention, meditation, delta, theta, low alpha, high alpha, low beta, high beta, low gamma, high gamma"

if (brain.update()) {
Serial.println(brain.readErrors());
Serial.println(brain.readCSV());
}
}

I used Roboduino--arduino-1.0.1

molavy, 2016-01-07:
i have exactly problem, 
when i connect t pin to tx i give one line no meaning characters,
when connect  t pin to rx i give no data
i using Thinkgear Am Development Start Kit

PCDotFan, 2016-02-02:
Me, too. I used Arduino UNO R3 and didn't see anything in Serial Monitor..

PCDotFan, 2016-02-11:
@molavy Ahh..It works, plz select "57600 Baud" and don't forget to replace "Serial.begin(9600)"

closed 2016-02-11 by kitschpatrol

ancymathai, 2016-04-03:
its not the serial data ......its only seen when reset button is pressed........

when replaced the arduino board problem solved