kitschpatrol/brain #18: Adding opto-isolater to mindflex

richardbeattie, 2016-05-08:
Hi all, I have a mindflex headset which I have set up according to this tutorial(http://www.frontiernerds.com/brain-hack) however the signal strength never reaches 0 and as such I cannot get the attention/meditation values. I tried adding salt water and that worked for a while but ultimately it stopped. So I purchased and opto-isolator(http://www.maplin.ie/p/high-voltage-opto-isolator-ay44x) but I am stuck at how I am meant to wire it up to the arduino and minflex. 

Any help would be greatly appreciated. 
Richard

ribas9521, 2017-05-13:
I would like to know too!

sixtyfive, 2019-02-15:
how would an optoisolator solve _that_ problem?

kitschpatrol, 2023-11-19:
Closing this as it's not related to a specific issue with the.

Perhaps an opto-isolator it could prevent noise between the Arduino and the TGAM, but given everything that's already going on right on the TGAM board, that seems a bit a bit superstitious.

closed 2023-11-19 by kitschpatrol