kitschpatrol/brain #19: TGAM module rev 2.7

penco193, 2016-11-23:
Hi!, i have a mindflex game with ThinkGear module rev 2.7. any of you can tell me if the firmware version is 1.6 or 1.7 for this board?, or where could i find it?. any help will be appreciated

faogustavo, 2017-10-17:
@penco193 did you solved this?

penco193, 2017-10-17:
Yes, thanks!

faogustavo, 2017-10-17:
Can you tell me how? I am with the same problem :/

sixtyfive, 2019-02-15:
fwiw, i have a `TGAM1_R2.8` that i've not yet been able to make work with `Brain`... not sure if the board revision is any kind of problem as it's still supposed to send the same data via the same pin, right?

3Dtj, 2019-03-01:
I seem to be having the same issue in getting data using the Brain sketch.  Will double check TX connection and library.

sixtyfive, 2019-03-02:
@3Dtj I ended up sort-of rolling my own as I couldn't figure out what was wrong. If you wanna give it a go: https://github.com/sixtyfive/tgam1

3Dtj, 2019-03-03:
Thanks brother.  I'll give it a go and see if it works out,

kitschpatrol, 2019-06-04:
Thanks — unfortunately I don't have access to the 2.7+ module to test against this library.

closed 2019-06-04 by kitschpatrol