kitschpatrol/brain #20: refresh rate of eeg power data

ciucianebbia, 2017-02-01:
The refresh rate of eeg power data can be increased ? 
I refer to the spectral values of the individual bands, not to meditation or attention.
1 Hz is toooooo slow.

Thanx

steeph-k, 2017-02-01:
It can't be changed by command or jumper, like the baud rate. The module is sending the data once every second. Much faster wouldn't really make sense for frequencies as low as 0.5 Hz. If you need more data points you will have to use the raw data, not the pre-calculated frequency bands.

kitschpatrol, 2017-02-01:
Thanks @steeph-k!

closed 2017-02-01 by kitschpatrol

ciucianebbia, 2017-02-04:
There is a code fragment to treat this raw data?
And i'ts also necessary to set the hardware on TGAM module ?  BR1 and BR2 to VCC and GND ?

Thanx

steeph-k, 2017-02-04:
I didn't find any project using the raw data when I searched a few years 
ago. Maybe you will find something helpful on http://developer.neurosky.com/

If your module is not pre-configured to send raw data, you don't 
necessarily have to change the jumper resistors. You can also switch the 
mode via command. See 
http://wearcam.org/ece516/neurosky_eeg_brainwave_chip_and_board_tgam1.pdf