kitschpatrol/brain #25: force trainer 2

RishabRao, 2019-12-29:
Does this work on the force trainer 2? Brain.update() keep returning false. readCSV gives 200, 0, 0

kitschpatrol, 2019-12-29:
Hi Rishab, I'm not sure since I don't have access to a Force Trainer 2.

Sounds like it should be possible according to [this review on Amazon](https://www.amazon.com/gp/customer-reviews/RUA60CMSE6K2Y/ref=cm_cr_arp_d_rvw_ttl?ie=UTF8&ASIN=B00X5CCDYQ).

Looks like you might want to try 57600 baud instead of 9600 when [establishing the serial connection](https://github.com/kitschpatrol/Brain/blob/master/examples/BrainBlinker/BrainBlinker.ino#L23).

I'd also suggest trying [this example](https://github.com/kitschpatrol/Brain/blob/master/examples/BrainSerialTest/BrainSerialTest.ino) and looking at the raw data coming in, this will tell you if you're getting any values a tall despite the bad connection, or whether the Brain library is running into other parsing errors.

Good luck and let me know if you get it working!

RishabRao, 2019-12-31:
Thank you so much for your response. It looks like switching to 57600 baud fixed the problem for the force trainer 2! Out of curiosity how did you know to use this rate? Thank you once again!

kitschpatrol, 2020-01-01:
Great! The 57600 baud tip was mentioned in the [Amazon review I linked](https://www.amazon.com/gp/customer-reviews/RUA60CMSE6K2Y/ref=cm_cr_arp_d_rvw_ttl?ie=UTF8&ASIN=B00X5CCDYQ).

It's also [one of the standard baud rates](https://en.wikipedia.org/wiki/Serial_port#Settings) you're likely to see serial connections use.

closed 2020-01-01 by kitschpatrol