kitschpatrol/brain #26: Nothing on Serial

beaverhacker, 2020-02-20:
I wired up everything correctly: the tx pin of my starwars force trainer II headset and the rx pin of my arduino uno. But the BrainSerialTest example does not write anything to the serial (which I made sure to be at baud 9600).

I was wondering if I broke my headset so i connected it to my phone and the hologram app worked fine.

I also read here: <https://github.com/JimRoskind/NeuroskyHacking/blob/master/ForceTrainer.ino>
that the headset should use a default baud rate of 57600 so I wrote an arduino script that relayed the raw serial data from it's rx pin to the serial monitor. This time, the serial displayed a stream of numbers so I assume there's nothing wrong with the hardware setup. As far as I can tell, the headset is meant to send 11 data values at a time, but I couldn't find any pattern in my results.

I found on an Arduino forum that someone else had the same issue but the issue just suddenly went away for them?

What can I do?

beaverhacker, 2020-02-20:
I tried out the code in the github I linked, and it's giving me something
[image]

but I'm confused on what it means and why the brain library doesn't work for me

zensutherland, 2020-02-20:
I'm also currently working on a MindFlex EEG project and found nothing output to the serial. 

I was getting no data from the BrainSerialTest example until i was a lot more careful about placement of the headset (sensor repositioned above the left eye) and used alcohol swabs on the sensors and the places where it met my face and ear lobes. Then i began getting data, and with more positioning finally a zero value in the brain.readSignalQuality() which indicates the best possible signal.

beaverhacker, 2020-02-20:
Isn't the code meant  to display signal quality 200 at the very least? I'm not getting anything displayed at all.

zensutherland, 2020-02-20:
I thought so too, but i'm not sure how the "if (brain.update())" operates, but i kept monkeying with the whole setup and positioning until i began getting data. I know that's not very scientific or logical, but that's pretty much all i've got.

beaverhacker, 2020-02-20:
Well thanks anyway, I'll try tomorrow.

zensutherland, 2020-02-29:
I was having trouble with serial connections and 32-bit processors, so i wrote this quick serial signal evaluator for when i'm using a Teensy 3.2. Maybe it'll show whether you're getting good transmissions from the MindFlex.

------------------------------------

#define HWSERIAL Serial2

int counter = 0;

void setup() {
  Serial.begin(9600);
  HWSERIAL.begin(9600);
  Serial.println("Mindflex test");
  Serial.println("-------------");
  Serial.println("Please Connect and Turn Turn on Headset");
}

void loop() {
  byte incomingByte;
  if (HWSERIAL.available() > 0) { // Anything coming in from Mindflex headset
    incomingByte = HWSERIAL.read();
    if (incomingByte == 0xAA) {  // First Header good, check next
      incomingByte = HWSERIAL.read();
      if (incomingByte == 0xAA) {  // Second Header good - Start Processing
        counter++;
        Serial.print("Header Signal Encountered (");
        Serial.print(counter);
        Serial.print(") ");
        delay(100);
        incomingByte = HWSERIAL.read();
        incomingByte = HWSERIAL.read();  // Read second byte
        incomingByte = HWSERIAL.read();  // Read Third byte
        Serial.print(incomingByte, DEC);
        if (incomingByte == 0x00) Serial.println(" - Signal Good");
        if (incomingByte > 0x00 and incomingByte < 0x20) Serial.println(" - Signal Marginal");
        if (incomingByte > 0x21 and incomingByte < 0xC8) Serial.println(" - Signal Very Poor");
        if (incomingByte == 0xC8) Serial.println(" - Headset Not Touching ");
      }
    }
  }
}

AdiSun123, 2022-02-10:
Hey Guys, I just started this up today, and got really disheartened by seeing nothing in serial pin. I tried a baud rate of 57600 too, changed some code and still got nothing. Is it just the placement of electrodes, positioning and set up? I checked my soldering and I don't think I've done anything wrong with the chip. Please help me with what I could try???

sixtyfive, 2022-02-10:
"keep upping" as in: start at 9600, try 19200, try 38400, then 57600, then 115200. sometimes there is a problem that only happens at higher rates, but not at lower ones. so yes, i suggest you start at low baud rates, but with [a simple TEST program](https://gist.github.com/sixtyfive/2ccb17fd1ace9f249b28c04255d0d6f3), not with BrainSerialOut. if you know that your test program works, then try again with BrainSerialOut.

AdiSun123, 2022-02-10:
Thank you so much for the help, I'll try it tonight and get back to you whether it work

AdiSun123, 2022-02-10:
Thank you so much for the help, I'll try it tonight and get back to you whether it works

AdiSun123, 2022-02-13:
So i began with trying to make changes in baudrate and I was getting quite consistent prints outside of the brain firmware. Then I also took some time to see if the voltage is flowing around the circuit using a multimeter checking every point from the rx and tx pins to the power to the TGAM and it all seems in place. I also tried removing the "if brain.update()" line of the script and that gave me 200, 0, 0 in output in each line of the serial monitor. Now I think there are only 2 possible issues but I still needed some help on that:

1. I haven't soldered 2 of the pins. For the TGAM, I have only soldered the ground and vcc pins which power the TGAM and not the ones that go to the Arduino. I thought that It mightn't be required but I'll give that a shot today, I've attached a photo github link; I have soldered the pins circled in red to a 2x aa battery pack for the required 3.3v power to tgam but I have left the other 2, in blue empty whereas on the sales sheet it shows that they should be going to the microprocessor (in this case the arduino). Is this why I'm getting nothing or only 200, 0, 0.
https://user-images.githubusercontent.com/99411573/153746485-87b45cb7-d344-4d48-bb6d-35917766e7fe.jpeg

2. Or I think it could be a possible issue with contact to skin. At the moment I'm just using AgCl dry electrodes and taping them without any sort of conductive gel or anything. Could that be the reason why I am getting no readings? 

Help would be really appreciated

steeph-k, 2022-02-13:
It's been years for me, too, since I've used a TGAM module. But as far as I remember you should get a reading every second. If you have no contact to the skin at all (or bad contact) you still get the '200' value.

And for soldering additional pins: Ground, Vcc and the two data wires are all that is needed. But there are bridges that you can short with a resistor to determine the default baud rate. Depending on where you got the module, there may already be a baud rate set (e.g. if you got it from a used toy that used something else than 9600). If I remember correctly, you can set the baud rate independently in software by sending a command over serial. But you have to know the baud rate that's set by the hardware bridges. Have a look at the datasheet for your version of the module to identify the bridges. But make sure to compare the version and revision printed on your board with the one in the datasheet PDF. I think there were tiny differences in later versions.

sixtyfive, 2022-02-13:
As for setting the baud rate in hardware as just mentioned by steeph-k: also see the README I mentioned, it has some info about that. Especially note that TGAM modules come with different firmware versions (1.6 or 1.7 if memory serves) and things work a little differently for either.

AdiSun123, 2022-04-10:
Hey, I'm so sorry, I just got really busy and left the project completely. 

[image]

Here is an image of the TGAM. So I've just soldered female jumper leads to the end of each of the connections and then connected each component to it.
**At the top:**
The black lead is connected to the electrode placed at the head.
The red lead is connected to a ear clip electrode
The black lead is connected to the other ear clip electrode
**At the bottom:**
The blue lead is connected to the ground of a 2x AA battery pack
The purple lead is connected to the positive of the 2x AA battery pack
The white Tx pin is going to Rx Pin (pin 0) on Arduino
The grey Rx pin is going to the Tx Pin (pin 1) on the Arduino

Despite the neat soldering and some conductive gel placed with the different electrodes (tried both AgCl and Steel and just about to try the gold plated ones) I am using the BrainSerialOut code, completely unchanged and still getting nothing in the serial monitor. And we have already verified that there is serial communication. And as for baudrate I tried both 9600 and 57600 and still haven't received anything.

What could I do now?

sixtyfive, 2022-04-10:
> The black lead is connected to the electrode placed at the head.
> The red lead is connected to a ear clip electrode
> The black lead is connected to the other ear clip electrode

I don't see two black leads.

But I do notice that you're not using shielded wires and we can't see how long they are. Both might be a problem.

Also I remember my Mattel toy needing 3x1.5V cells in series, not 2x. That's 4.5V, which is almost 5 - and I can't remember seeing anything like an LDO on the carrier PCB. Have you checked the TGAM datasheet and is it indeed happy with just 3V?

sixtyfive, 2022-04-11:
Alright, in that case the Mattel Mindflex carrier board must after all have a little LDO to get down from it's 4V5 to the 3V3-ish required by the daughter board. And as @steeph-k pointed out, you should still be getting 200s back when there's no data to report. I'm sorry, I'm drawing a blank here at the moment ... if jumpers are set correctly and baud rates match then I don't know. In about two weeks I'll be able to go by the workshop again and I think I still have the Mindflex there and might give it a go again.

AdiSun123, 2022-04-11:
Just another thought, I think I know why I'm not getting the 200,0,0. It might be because my arduino code starts with the condition if brain.update() or if serial.available() > 0, and I am assuming that I am getting no serial data on rx, tx and hence the function isn't fulfilled and the code doesn't give any result on the serial monitor. When I removed this condition then I start receiving 200,0,0. Thank you so much for your help and If you get any ideas while going through the workshop your further help will be greatly appreciated.

abdushams, 2022-11-09:
Hello guys, 
i've connected the mindflex same as in this write up  https://frontiernerds.com/brain-hack and tried the BrainSerialOut example and i got nothing on the monitor .. also tried the steps and the code above but nothing worked .. also noticed that the RX LED on the arduino doesn't blink at all so i think the problem is that the arduino is not receiving data from the headset  .. any idea on how to solve that ??

kitschpatrol, 2023-11-19:
Thanks for this conversation. Getting a solid connection is indeed tricky. Closing since it's not an issue specific to the library.

closed 2023-11-19 by kitschpatrol