kitschpatrol/brain #28: Reading Serial Data Manually

nikhilAnayak, 2020-05-13:
I couldn't figure out how to make this library work and was wondering how I could parse serial data manually. I am getting data but I don't know what it means. Any help would be appreciated.

sixtyfive, 2020-05-13:
Parsing the serial data directly is tough work. Perhaps it'll help you to read through https://github.com/sixtyfive/tgam1/tree/master/lib/tg_stream_parser ?

nikhilAnayak, 2020-05-13:
> Parsing the serial data directly is tough work. Perhaps it'll help you to read through https://github.com/sixtyfive/tgam1/tree/master/lib/tg_stream_parser ?

This seems more general. I have data and am wondering what it means in the context of an EEG device.

sixtyfive, 2020-05-13:
Looks like it might be the signal of one channel, as in, the digitized version of how many mV were measured by that channel's electrode relative to the ground or reference electrode at those moments in time. You might want to turn that into an x-y-plot or perhaps do some FFT (fast fourier transform) on it, like for example @kitschpatrol's other project, https://github.com/kitschpatrol/BrainGrapher, does.

nikhilAnayak, 2020-05-13:
Ok, thanks. From the data I have, how would I do something like getting the attention value or something similar?

sixtyfive, 2020-05-14:
Hmm. I only _guessed_ that what you're seeing is raw data. That's not certain - you never gave any information about _how_ you get that data. If indeed it is and instead you're interested in the attention value, then you might have to change the way you read out the chip. The attention value is calculated internally and send out as part of the unaltered serial output for any of the commercial devices I'm aware of. Also be aware that noone knows what the chip does to calculate either the attention or the meditation values and there's criticism of that. Others (like me) are interested in the EEG power bands only and prefer to calculate those themselves from the raw data.

nikhilAnayak, 2020-05-14:
I got this data by connecting a wire from the "T" pin of the chip to the serial "RX" pin. I then directly read the input serial data. This is what I've seen people doing but the Brain library didn't work after doing this so I was wondering if I could get data myself.

sixtyfive, 2020-05-14:
Ah, in that case it's more likely you have the preprocessed data (unless you also played with the resistors on the TGAM board). Try my project (https://github.com/sixtyfive/tgam1; make sure you go through the config file!) instead of the Brain library, see if you can get to work. Perhaps if that works you'll have an easier time figuring out how to get the Brain library to work.

nikhilAnayak, 2020-05-14:
Is there another way I can talk to you(discord or something else) because I have a few other questions?

sixtyfive, 2020-05-14:
Sorry, I stay off of social networks for precisely those reasons. One place where you should be able to find more help is https://electronics.stackexchange.com. If you want to help the people there to help you, also have a look at https://stackoverflow.com/help/how-to-ask.

nikhilAnayak, 2020-05-14:
Ok, that makes sense. Thanks a lot though.

nikhilAnayak, 2020-05-14:
Also, with the tgam library, I'm getting a lot of errors with the different libraries: #include <ESP8266WiFi.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>
WiFiClient wifi_client;

#include <PubSubClient.h>

I can't find ones that work. Where should I get them from?

nikhilAnayak, 2020-05-15:
One last thing. Reading serial data from the 57600 baudrate gives me this data:
https://pastebin.com/CPxZ4mTu

Does this have any significance? I briefly looked at the neurosky documentation but didn't see anything.

sixtyfive, 2020-05-15:
It's just numbers. Without knowing what resistors you have set on your TGAM board I don't think it's knowable what they're meaning is. Also, if I remember correctly (this is all more than a year ago for me – Read The Source, Luke!) the TGAM serial data doesn't include newlines. Certainly doesn't include timestamps. So where are those newlines coming from and how do you know they're not arbitrary?

steeph-k, 2020-05-15:
Short answer: The numbers you receive at 57600 contain the raw value. It's every 8th number in the stream. You can identify it by looking for the number 170 being sent twice in a row. The number before that was the raw EEG value. The number after the double 170 is also important. It indicates how good the electrode connection is. All other numbers represent calculated values that in my experience have no practical use.

But this is just my guess from looking at the numbers. If you want to be sure I'd have to know which version of the TGAM you are using and what the settings are.

I hope that helps. If not, or if you're interested in a longer answer, let me know.

nikhilAnayak, 2020-05-15:
So what does the 8th number represent? You said "It's every 8th number in the stream." but what is "it"? And how could I get the attention/meditation values from this?

(Sorry if this is a dumb question, I'm just trying to figure out how this works)

sixtyfive, 2020-05-15:
By "calculated values", I understand @steeph-k to mean the attention and meditation values.

nikhilAnayak, 2020-05-15:
Ohhhh. Thanks a lot both of you guys.

steeph-k, 2020-05-15:
Well, I should have said the 7th and 8th byte. It's a 2-byte-number. But yes, it's the raw value that you would perform FFT on or feed to some software to interpret it. Keep in mind I'm just guessing as to the settings of your module. But the "eSense Attention" value, as Neurosky calls it, should be the second number after the 170 170 and "eSense Meditation" the third.

nikhilAnayak, 2020-05-15:
On the neurosky documentation, it says esense attention is 0-100 but in the second Pastebin you can see that sometimes the number after the 170, 170 is 255. What does this mean?

nikhilAnayak, 2020-05-15:
Actually, these are all the "attention" values from the pastebin:

[128, 128, 128, 128, 128, 128, 128, 128, 128, 128, 128, 26, 170, 170, 255, 170, 170, 170, 170, 170, 170, 133, 170, 170, 119, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 32, 170, 170, 170, 170, 119, 170, 170, 119, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 112, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 133, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 119, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 119, 
170, 170, 119, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 119, 170, 22, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 119, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 119, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 133, 170, 170, 133, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 119, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 252, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 119, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 119, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 
170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 249, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 119, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 119, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 119, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 170, 119, 170, 119, 170, 170, 170, 170, 133, 119, 170, 170, 170]

They are all similar which makes it seem less likely that these are the actual meditation values.

steeph-k, 2020-05-15:
It means that I didn't take another look at your output before writing my last comment and that my guess was wrong.

nikhilAnayak, 2020-05-15:
lol. Well, no problem. I'll just look over the data again and see if there's anything else that I find.

steeph-k, 2020-05-15:
Well, most of that output seems unusable anyway. It looks like there are many numbers missing, most likely because of a bad connection for the serial transmission. If you're only after the eSense values, you could switch to 9600 baud anyway. This makes transmission errors much less likely and setting the mode ensures you know what the module actually sends. If you have a soldering iron with a small tip you can set the bridges accordingly. If I remember correctly you can do the same by sending a command code at 57600 baud, then switch to 9600 on your receiving device. It sound like you have the spec sheet that explains the codes. I'd have to look them up myself.

nikhilAnayak, 2020-05-15:
Ok. Is there any useful data in the 9600 baud serial output(https://pastebin.com/vyYLFxBW)? I'm just not really sure what to be looking for.

steeph-k, 2020-05-15:
I'm not sure there is useful data in this output. Are you sure your serial connection is okay and without strong interference? What are you using to read the data? The Brain library worked well for this when I used it. Maybe have another go at adding the library to your Arduino?

nikhilAnayak, 2020-05-15:
When I used the library, it only prints the data when brain.update() is true, which never happened. I don't really know why that's happening.

nikhilAnayak, 2020-05-15:
Also, I'm just reading the raw serial data at the 9600 baud rate.

steeph-k, 2020-05-16:
Does the module send at 9600? Because the 57600 stream looked right (although incomplete). You have to set it to "normal" mode before you read at 9600. Other than that the only idea I have is to check the serial connection and make sure the receiver works probably. Maybe try a different software there as well. Because the output at pastebin is not raw. It has timestamps added. Maybe that's not all your receiver changes.

sixtyfive, 2020-05-16:
i'll note that i had to put full grounded sheet metal shielding around my board before the raw data values became usable and the signal began resembling actual EEG data. and yes, 9k6 baud will be easier to manage if you're not interested in the raw data anyways.

fwiw, what _didn't_ work with my board was to tell it in software to switch to 57k6 when the resistors were untouched. but after removing that resistor that needed removing and then running at 57k6, using software to tell it to go back down to 9k6, that worked fine.

steeph-k, 2020-05-16:
That was my experience as well.

nikhilAnayak, 2020-05-16:
What do you mean put full grounded sheet metal around my board? Do you have a picture or something I could look at?

nikhilAnayak, 2020-05-16:
When I run at the 57600, I get theoretically good data(following the payload conventions in the neurosky documentation) but the data seems basically unusable. Is that what you mean?

sixtyfive, 2020-05-17:
I mean this kind of shielding https://i.stack.imgur.com/iZFxX.jpg (you can get the metal by taking apart anything that has HF circuits inside; an old TV from the side of the street usually does the trick; depending on what tools you have available you might want to get a fitter or locksmith to help you with cutting and bending it into a new shape. Note that it will do jack shit if it's not grounded). Your "theoretically good data" is exactly that: it is _theoretically_ good, but _practically_ unusable because, one way or the other, it's just amplified and digitized noise.

nikhilAnayak, 2020-05-17:
Shouldn’t all that be unnecessary if I’m using a wire through the serial port rather than Bluetooth?

nikhilAnayak, 2020-05-18:
Also, I'm using a few inches of aluminum wire.

sixtyfive, 2020-05-18:
Aluminum wire? That's a little strange in and of itself, but okay. And no: each little wire is also a little antenna. And this is a very basic serial protocol here, there's no safety measures for making sure what was received is indeed what was sent. Got a cellphone in your pocket? Some kind of room lighting overhead? Mains wires running through the walls? A person standing next to the device, or perhaps, the device is even physically touching a person? Because each person is also a huge antenna again. But sure, the bigger problem is likely to be shielding of the analog portion of the circuit rather than the digital portion. Also because those wires you're using to transmit (digital!) serial communications are transmission antennae, too. Don't underestimate the detrimental effects of any EM field on something as weak as an EEG signal. The noise can very easily reach a level at which it completely overpowers any signal and worst case, you could be none the wiser.

nikhilAnayak, 2020-05-18:
Would changing the wire help with this problem?

sixtyfive, 2020-05-18:
I have no clue how long your wire is, whether it has twisted strands, what kind of shielding, if any, there is, what diameter it is, and indeed, even though I suspect it not to matter much, what the properties of aluminum are wrt absorbing or emitting EM fields.

A picture of the whole setup might help?

nikhilAnayak, 2020-05-18:
[image]
This the outside of the setup. I put the cover back on so you can’t see the neurosky chip directly. The blue wire is serial and the white is ground.

sixtyfive, 2020-05-18:
Okay, so it would be interesting to know where inside that black plastic box the TGAM board sits. Your Uno is mostly digital circuitry, so further away would be better. As for the cables, try twisting them around one another (something like what can be seen here: https://upload.wikimedia.org/wikipedia/commons/thumb/c/cb/UTP_cable.jpg/220px-UTP_cable.jpg). Also, what are you powering the Uno with (I don't see anything going from that battery to the Uno)?

nikhilAnayak, 2020-05-19:
[image]

This is a picture of the inside(this is just the standard Star Wars force trainer II toy. I’m planning to make the wires copper and move the Arduino further away. I’ll tell you if this works.

sixtyfive, 2020-05-19:
Ah, interesting, I've never seen one - only own the MindFlex one myself. Yours looks like much better hardware design. So, your wires are just standard dupont breadboarding wires, don't bother, they'll do fine. In addition to making them into a twisted pair, you want to do one other thing: you're currently getting GND off of the battery which creates a large-ish ground loop. The TGAM has a GND pad right next to the serial pad. I forgot which one it is but you'll find that in the datasheet. Attach your GND wire _there_. These two measures together (twisted pair and different location for GND connection) might already help a lot. Next, try what happens when the Arduino is not mechanically attached to the casing like in your first picture, but being held away from Force Trainer as far as your two cables will allow. Then, for attaching it mechanically, try to put a thing metal plate between Uno and Force Trainer (insulate the Uno's bottom first!) and see if that prevents the signal from worsening too much.

nikhilAnayak, 2020-05-19:
[image]
Is it the P3 Pin 1(the one on the bottom right of the top board)?

sixtyfive, 2020-05-19:
I would say so, yes. If you want a certain answer, though, it'll have to wait until the weekend when I'll have time to go by the workshop and check. This whole Corona situation has taken up any free time that was normally there during the week, sorry.

nikhilAnayak, 2020-05-19:
Yeah, it's fine. I think that's the right one. I'll tell you if the data is more readable now.

nikhilAnayak, 2020-05-19:
I'm still getting this "unreadable" data after changing the wires and using a different ground. I'm not sure what to do now because I feel like I've tried everything. If you have any ideas that would be appreciated.

sixtyfive, 2020-05-20:
Hmm. Does it still appear to be working in its original capacity?

nikhilAnayak, 2020-05-20:
Yeah, I’m still getting data. I never got to do a baseline test though. There should be an app you can test with for the Star Wars force trainer but it wasn’t available.

sixtyfive, 2020-05-20:
This one here? https://play.google.com/store/apps/details?id=com.unclemilton.ftII&hl=en Says something in the comments about problems with screen rotation. But okay, so it seems to not be broken. What path did you choose for changing the baud rate? Just software, or was that done via hardware modification?

nikhilAnayak, 2020-05-20:
Yeah. I have an iPhone/iPad and the app is not there. I also have an old android tablet and it is unsupported.
Also, I used software to change the baudrate.

sixtyfive, 2020-05-20:
>  Also, I used software to change the baudrate.

In that case, what do the numbers look like at 9k6Baud?

nikhilAnayak, 2020-05-20:
10:41:31.498 -> 170
10:41:32.527 -> 170
10:41:33.506 -> 4
10:41:34.533 -> 128
10:41:35.520 -> 2
10:41:36.550 -> 0
10:41:37.535 -> 124
10:41:38.522 -> 1
10:41:39.552 -> 170
10:41:40.532 -> 170
10:41:41.556 -> 4
10:41:42.537 -> 128
10:41:43.562 -> 2
10:41:44.548 -> 0
10:41:45.528 -> 138
10:41:46.550 -> 243
10:41:47.578 -> 170
10:41:48.561 -> 170
10:41:49.537 -> 4
10:41:50.565 -> 128
10:41:51.588 -> 2
10:41:52.567 -> 0
10:41:53.546 -> 135
10:41:54.574 -> 246
10:41:55.554 -> 170
10:41:56.580 -> 170
10:41:57.559 -> 4
10:41:58.591 -> 128
10:41:59.569 -> 2
10:42:00.598 -> 0
10:42:01.582 -> 123
10:42:02.610 -> 2
10:42:03.592 -> 170
10:42:04.574 -> 170
10:42:05.601 -> 4
10:42:06.583 -> 128
10:42:07.607 -> 2
10:42:08.593 -> 0
10:42:09.616 -> 116
10:42:10.598 -> 9
10:42:11.623 -> 170
10:42:12.604 -> 170
10:42:13.630 -> 4
10:42:14.608 -> 128
10:42:15.633 -> 2
10:42:16.613 -> 0
10:42:17.642 -> 118
10:42:18.621 -> 7
10:42:19.645 -> 170
10:42:20.625 -> 170
10:42:21.607 -> 4
10:42:22.632 -> 128
10:42:23.611 -> 2
10:42:24.637 -> 0
10:42:25.620 -> 117
10:42:26.648 -> 8

sixtyfive, 2020-05-20:
Okay, and that sequence is rejected by the Brain library? How are you feeding it into there?

nikhilAnayak, 2020-05-20:
I'm not using the Brain library right now. The Brain class has an update function(boolean) which seems to always return false.

sixtyfive, 2020-05-20:
Do feed that stream into something that's built for reading the TGAM serial data, try to get _some_ visual representation. Looking at just those numbers that way might look like it's bogus and not actually be bogus at all. I'll point out that there's a collection of links in my [tgam1 README.md](https://github.com/sixtyfive/tgam1/blob/master/README.md).

nikhilAnayak, 2020-05-20:
Ok. I'm going to try using kitschpatrol's BrainGrapher because I already have Processing set up which makes it easier. Should I just print the serial data that I'm receiving(like I'm doing right now)?

sixtyfive, 2020-05-20:
Follow http://www.frontiernerds.com/brain-hack I guess. Worked fine for me when I tried it back then.

nikhilAnayak, 2020-05-21:
That website uses the Arduino Brain library which doesn't seem to be working for me.

steeph-k, 2020-05-21:
But the last stream you posted now looks good to me. That update function only returns true if the received packet of numbers was parsed successfully. So I would have expected it to always return false with the last 9600 baud stream you have posted. I think it's worth trying again now.

nikhilAnayak, 2020-05-21:
That was 57600 baudrate. I just thought there were two sync(0xAA or 70), then a packet length, then the attention value. Following this the attention value was always 128.

[3 more replies]

closed 2023-11-19 by kitschpatrol