kitschpatrol/brain #30: Question about the Frequency

Revix1998, 2022-03-24:
hello, im trying to do a project where i can read the Delta, alpha, beta ... ect.

my probelm is that i dont know how to pull the frequency as an output. 

thank you

steeph-k, 2022-03-24:
The TGAM module uses its own hidden algorithm to decide which frequency range currently dominated the signal. So in this simple mode you only get values for the ranges. It can be switched into a mode where it additionally outputs filtered but otherwise raw data. You could use this stream to do your own calculations. But there will always be several different frequencies contained in a brain wave pattern.

steeph-k, 2022-03-24:
Sorry, I think I completely misread your question. If you want to switch LEDs on/off, this is not directly related to the TGAM module or this project. Do you want an LED to reflect the raw data? Or some other frequency? Maybe you could use PWM.

Revix1998, 2022-03-25:
yes using PWM , but my main problem is that how can i separate each value through a single LED, 

for example if i wanted to show  the code (unsigned long readMidGamma();) on a single LED is that possible ?

sixtyfive, 2022-03-25:
Maybe consider using a [Neopixel](https://www.adafruit.com/product/1734) and make it more red for a lower value and more green for a higher value, or some other colours...?

kitschpatrol, 2023-11-19:
Thanks — closing this since it's not actually an issue related to the project.

closed 2023-11-19 by kitschpatrol