kitschpatrol/brain #31: EEG power bands ?

Revix1998, 2022-05-21:
hi , 

i wanted to know if i can use the power band frequency as an output , 

i used  brain.readDelta .. .. and so on but still i have nothing on the LED 

is there something i have to do first before ? considering am new to programming 

i link a portion of my code so  its clear 

#include <Brain.h>
Brain brain(Serial);
 int whitePin = 13;
 int yellowPin = 12;
 int orangePin = 11;
 int BPin = 10;
 int white2Pin = 9;
 int greenPin = 8;
 int redPin = 7;
 int orange2Pin = 6;
long interval = 500; 
long previousMillis = 0;
int ledState = LOW; 
int HighAlpha;
int Delta;
int MidGamma;
int LowGamma;
int Theta;
int LowAlpha;
int LowBeta;
int HighBeta;
 unsigned long readDelta();
 unsigned long readHighAlpha();
 unsigned long readLowAlpha();
 unsigned long* readPowerArray();
 unsigned long readLowGamma();
 unsigned long readMidGamma();
 unsigned long readTheta();
 unsigned long readLowBeta();
 unsigned long readHighBeta();
void setup() {
 // Set up the LED pin.
 pinMode(greenPin, OUTPUT);
 pinMode(yellowPin, OUTPUT);
 pinMode(redPin, OUTPUT);
 pinMode(whitePin, OUTPUT);
 pinMode(white2Pin, OUTPUT);
 pinMode(BPin, OUTPUT);
 pinMode(orangePin, OUTPUT);
 pinMode(orange2Pin, OUTPUT);
 
 
 // Start the hardware serial.
 Serial.begin(57600);
}
void loop() {
 // Expect packets about once per second.
 if (brain.update()) {
 Serial.println(brain.readCSV());
 
 // Power bands .
 Delta = brain.readDelta();
 HighAlpha = brain.readHighAlpha();
 MidGamma = brain.readMidGamma();
 Theta = brain.readTheta();
 LowAlpha = brain.readLowAlpha();
 LowBeta = brain.readLowBeta();
 HighBeta = brain.readHighBeta();
 LowGamma = brain.readLowGamma();
 }
 
 
 if( brain.readSignalQuality () == 0)

kitschpatrol, 2023-11-19:
Which headset are you using? Not all headsets provide EEG power band data, as described in a table from the [tutorial page](https://frontiernerds.com/brain-hack) pasted below:

[image]

kitschpatrol, 2023-11-19:
Closing pending confirmation of the headset you're using, but in general no additional configuration should be required to get EEG power band values from a compatible headset.

closed 2023-11-19 by kitschpatrol