kitschpatrol/brain #37: Serial data comes in the format 49, 50, 49

sukimaruca, 2025-06-29:
Good evening, afternoon or day to whoever is reading this.

To give you some context, I am using a TGAM1 module, which contains two clips and an electrode for the forehead.

All connections are working correctly, but the serial values ​​sent are not what I expected:

Quality - 49

Attention - 50

Meditation - 49.

CSV - 49,50,49

I tried everything to check what was wrong, from unsoldering the board to check without the clips to adding resistors from the Arduino RX to the master Bluetooth.

Connection:

Tgam1 > JDY30 transmits to an HC-05 that sends the data to the serial.

Code below:

# include <Brain.h>
# include <SoftwareSerial.h>

// Configurar pinos para comunicação com o TGAM
SoftwareSerial tgamSerial(8, 9); // RX, TX

// Criar objeto Brain
Brain brain(tgamSerial);

void setup() {
  // Iniciar comunicação serial com o computador
  Serial.begin(9600);
  // Iniciar comunicação com o TGAM
  tgamSerial.begin(9600);
  Serial.println("Iniciando leitura do TGAM com biblioteca Brain...");
}

void loop() {
  // Verificar se há novos dados do TGAM
  if (brain.update()) {
    // Exibir qualidade do sinal
    Serial.print("Qualidade do Sinal: ");
    Serial.print(brain.readSignalQuality());

    // Exibir nível de atenção
    Serial.print(" | Atenção: ");
    Serial.print(brain.readAttention());

    // Exibir nível de meditação
    Serial.print(" | Meditação: ");
    Serial.println(brain.readMeditation());

    //CSV
    Serial.print("CSV: ");
    Serial.println(brain.readCSV());
  }
}

kitschpatrol, 2025-06-30:
Can you clarify what serial values you were expecting? The format you're showing is correct — but the first CSV value, `49`, indicates an issue with the signal quality. (A value of `0` would indicate a good signal, `200` indicates no signal at all.)

Signal connection issues are tricky. You should double-check the placement of the electrode, the continuity of the ground clips, and potential noise from your power sources. I'd also recommend removing as many variables as possible and then increasing the complexity of your circuit once a good `0` connection can be confirmed (e.g. use hardware serial, direct connection to the serial pads on the TGAM to your .)

There's a small chance your TGAM module is configured for a different Serial bit rate, or you might want to set the 50Hz filter instead of the 60Hz filter depending on where you're located. (These are set via jumpers on the TGAM board, see [the module's documentation](https://cdn.sparkfun.com/datasheets/Sensors/Biometric/tgam1.pdf) for details, specifically the "Configurable Default Settings".

If you're issue isn't with the signal quality but rather that you were expecting additional data in the CSV like the EEG power band values, then this is also related to and potentially remedied by the board's configuration as outlined in the document linked above.

sukimaruca, 2025-06-30:
Hello, good evening.
When I mentioned what I expected from the signals, I expected the following CSV sequence:
Quality, Attention, Meditation, Delta, Theta, etc. But only these three signals arrive, and they are wrong.
Why are they wrong? Because these three values ​​appear even without the users using the electrodes and clips.
As for the electrodes, I have already checked the grounding and continuity, everything is fine.